instamatic-dev/instamatic #16: Installing via conda

thomasaarholt, 2019-11-19:
Hi Stef,
I'm installing instamatic to test on our Titan G2 60-300 here in Oslo (in an anaconda environment).
In addition to performing some electron diffraction experiments, I was hoping to take advantage of the programmatic control of the microscope to acquire EELS spectra. Not spectrum images directly, but eels spectra as a function of shifting the diffraction pattern over the GIF. Is there any way to do that with instamatic in its current form?

I'm super impressed with the software, btw. We had a demo by Hongyi Xu last week.

thomasaarholt, 2019-11-19:
Sorry about the change of title. I originally started writing a different issue as I was unsure if I'd be able to install Instamatic in a Ananconda environment on an offline microscope PC, but that seems to have succeeded (checking properly later this week).

stefsmeets, 2019-11-19:
> eels spectra as a function of shifting the diffraction pattern over the GIF

Hi Thomas, if you mean via some kind of shift via the projector or diffraction alignment coils, then this is quite straightforward to do via for example `ctrl.diffshift`. Then you could read out the detector using the camera interface in instamatic.

I recently included some code to run DM scripts from Python (https://github.com/stefsmeets/instamatic/blob/master/instamatic/camera/gatansocket3.py) that I intend to use for communication with DM that you may want to have a look at. I haven't had the time to integrate it with the rest of the code yet, but it may give you some more flexibility.

Btw, do let me know how it is working for you. I only have JEOLs to test on, so I would be happy with any feedback you have using instamatic on a Titan!

thomasaarholt, 2019-11-19:
This is great! I noticed the ctrl.diffshift, and what you propose is exactly the plan!
I've just created #17 to track difficulties I may have with setting up instamatic on our Titan.

stefsmeets, 2019-11-19:
> I've just created #17 to track difficulties I may have with setting up instamatic on our Titan.

Thanks for doing that, I will keep an eye on it!

closed 2019-12-17 by thomasaarholt