instamatic-dev/instamatic #18: Installing Instamatic on an FEI Titan G2

thomasaarholt, 2019-11-19:
I'm following `config.md` to set up instamatic to talk to our Titan. I did the following to set up my Anaconda environment on the offline Titan computer:

### Setup
- Installed Anaconda on an online PC.
- Created a new environment, and installed all the dependencies needed for instamatic through conda/pip*, but not instamatic itself. (I originally did install instamatic but on the Titan I got a path error from the `instamatic` script due to different user names on the Windows user)
- Installed Anaconda on the offline PC
- Copied over the environment folder from `$HOME\Anaconda3\envs\`
- Called `pip install instamatic-master.zip` on the Titan pc.

### Launching Instamatic
- I notice that I need to launch `instamatic` (from cmd) twice the first time its used - the first time it creates a certain file but then raises an error. On the second time it runs smoothly and opens a simulated environment GUI with a high fps screen.
- I ran `instamatic.autoconfig.exe` but end up with the following [error.txt](https://github.com/stefsmeets/instamatic/files/3863586/error.txt).
- I fixed the script error by modifying the python file, but I now observe that only the Orius camera is supported. (see attached [error2.txt](https://github.com/stefsmeets/instamatic/files/3863650/error2.txt))

We have two Ultrascan cameras on our system (models US1000XP and US1000FTXP). They are identical cameras with different shutters. Is there any hope of adding support for those?
We do have an Orius camera on our JEOL2100F, but that's of slightly less interest to me.

thomasaarholt, 2019-11-19:
I should mention - I did try to choose the Orius camera. Then there's a blank Windows popup dialogue that appears followed by the error in error2.txt.

thomasaarholt, 2019-11-19:
Okay, so I've realised that I should probably just skip the autoconfig, since I now see that templates for things like the `fei` microscope are actually named `fei_themisz`.
[1 min later]:
Hehe, I think I just got it to work, will report some errors in separate issue.

thomasaarholt, 2019-11-19:
Current issue: The fei_microscope doesn't have a getCurrentDensity method. I can create a dummy `pass` method for now and see if that works, but I suspect there'll be a few more of the same sort. Could you take a look? I need to pass on the microscope to the next user now, but I'll be back on tomorrow morning. (CET timezone here)

stefsmeets, 2019-11-19:
> Could you take a look?

Yeah, I will do so tomorrow. 

The error you are getting is because are probably missing the DM plugin. I'm using the one from RED, you can get it from here: https://zenodo.org/record/2545322#.XEW5sC0o8uU (`REDc.rar` -> `CCD_plugins` -> `REDCCDPlugin.dll`). It should go in the DM plugins directory (`C:\ProgramData\Gatan\Plugins`). Hope that works!

thomasaarholt, 2019-11-20:
Yep, that did the trick. I created a dummy `getCurrentDensity` method for the `fei_microscope`, and now the instamatic GUI opens.

I just tried doing a cRED acquisition, but I do get an [error](https://github.com/stefsmeets/instamatic/files/3868066/error3.txt) here as well once it attempts to read from the gatan camera. I tried with and without live view on in GMS, with both our cameras (This raises the question: How does instamatic know which camera to acquire from?)

thomasaarholt, 2019-11-20:
Btw, I've got full control of the beam aspects. Really enjoying making the beam do my bidding by spinning in spirals :)

thomasaarholt, 2019-11-21:
I got super quick help from the SerialEM community, and installed it this afternoon. I was shortly able to both successfully run the `gatansocket3.py` "hello world" printing in GMS3, as well as acquire images from the ctrl.getRawImage() function! I'm not sure if getRawImage depends on SerialEM? In any case, this is very exciting, and I'm looking forward to seeing what I can get out of this, particularly with respect to EELS!

stefsmeets, 2019-11-22:
> I got super quick help from the SerialEM community, and installed it this afternoon. I was shortly able to both successfully run the `gatansocket3.py` "hello world" printing in GMS3, as well as [...]

That's nice to hear! The `getRawImage` uses the `RED` plugin for DM I linked before. It was developed in Stockholm and therefore easy to get started with.

The `gatansocket3.py` uses the SerialEM plugin for DM and is much more flexible and better maintained. I came across it a while ago and I was curious whether it could be used to do continuous acquisition through Python in DM. I find it particularly exciting that you can send execute DM scripts, something I wish I had a couple of years ago. I intend to make `getRawImage`/`getImage` depend on that, as well as continuous acquisition, but I have not gotten to it yet.

closed 2020-04-03 by stefsmeets