instamatic-dev/instamatic #20: FEI microscope file error?

thomasaarholt, 2019-11-19:
Looks like the following input code doesn't return into anything.
https://github.com/stefsmeets/instamatic/blob/688266cd61d331fd6f822b0c4606bfbf93474924/instamatic/TEMController/FEI_microscope.py#L175

asdfdsa, 2019-11-19:
I may have a little comment here: the line just urges the user to confirm the type of stage before continue (i.e. press the little "enter" icon on the FEI software), otherwise the API crashes for some reason. At least that was my impression from like a year ago

thomasaarholt, 2019-11-19:
Oh! You mean the "choose Single/Double Tilt" holder + Connect the holder cable thing?

Btw, I'm jealous of your username. Must be the most satisfaction one can have when logging in.

asdfdsa, 2019-11-19:
Exactly.
(Regarding the username, I actually wanted to change it to my real name but apparently I risk disassociating all my current repositories... Registered like 6 years ago without thinking too much)

closed 2019-11-20 by asdfdsa