instamatic-dev/instamatic #21: FEI Diffshift comments

thomasaarholt, 2019-11-20:
Looks like diffshift is limited to between -1 and 1 unnecessarily. I disabled the check on our Titan, and it happily goes much further out.
https://github.com/stefsmeets/instamatic/blob/e3b091bba82b53e84ac0e9ec5b14ede731a72335/instamatic/TEMController/fei_microscope.py#L462

Also, how come the FEI diffshift uses radian units? @asdfdsa Not complaining, just curious.
https://github.com/stefsmeets/instamatic/blob/e3b091bba82b53e84ac0e9ec5b14ede731a72335/instamatic/TEMController/fei_microscope.py#L467

asdfdsa, 2019-11-20:
It actually took me some time to figure out the way how diffshift is encoded in the FEI system. I also don't understand why they want to use radian units, but it turned out to be like that. 

At first I just took it into granted that every lens is encoded with limits from -1 to 1, and that was why I added the -1 and 1 check. But good to know that it is not limited to -1 and 1! But if you have time please check the limit, I am pretty sure there is a limit because it does not respond if you give it a too high value (I knew that because at first I was not doing the conversion of radians and passed some number with which the FEI system was not responding. I guess the number was out of the limit)

thomasaarholt, 2019-11-20:
@asdfdsa Might be an obvious question, but do you know how to control diffraction shift ("User Diffraction Shift") on the actual microscope? With TEM User Interface we have access to the "Diffraction Alignment" (terrible naming), but that is a bit different in that it sets the "zero" for the diffraction shift.

thomasaarholt, 2019-11-20:
The maximum value (at least on our system) is determined by the sum of the value of the user diff shift and the alignment diff shift. Their sum must be between -1 and 1 in "TIA units".

With the alignment diff shift set to 0, I can set the Instamatic diff shift values between -28.6 and +28.6. In rad, this becomes +- 0.5.

So ideally the system would check the alignment diff shift as well, and the magnitude of their sum should not go above 0.5 rad. (not sure why I'm off by a factor of two compared to your experience)

asdfdsa, 2019-11-20:
On the actual TEM I remember that one needs to map the diffshift function to the multifunction knob first and then get the access to diffraction shift.

+-1 is not my experience but just my (invalid) assumption. Looking at the API, I don't think I interfaced the "align diff shift", but I feel it should be possible to interface.

closed 2019-11-24 by asdfdsa