instamatic-dev/instamatic #24: fei_microscope missing getMagnificationRanges

thomasaarholt, 2019-12-16:
Looks like there was one more thing needed to get the `fei_microscope` working, which was adding a `getMagnificationRanges` method. Currently I've just made it return an empty dict, but I'm sure there's a better return value.

stefsmeets, 2019-12-17:
Good point! I will have a look at it.

stefsmeets, 2019-12-17:
Can you tell me when this happened? Because it seems like this is only accessed when doing the `autoconfig`.

thomasaarholt, 2019-12-17:
Ah yes, that makes sense. Since I was having trouble (in the other issue), I decided to reinstall everything after your merged PR yesterday(install from source). I used the autoconfig to set things up. Can't quite remember the exact details, but what you are describing makes sense.

commit 79055b9c (2019-12-19): Raise and handle error when magnification ranges are requested. Fixes #24.
closed 2019-12-19 by stefsmeets