instamatic-dev/instamatic #32: Controlling multifunction knobs

thomasaarholt, 2020-04-22:
Hi Stef!
I see you've been busy with instamatic. Looking forward to having a look through what is new.

These days we're trying to use the microscope remotely as much as possible. The biggest challenge right now is doing basic tuning remotely without the control panels we normally have.

Is it possible to access the multifunction knobs using the instamatic python interface? I realise that in general, instamatic and serialEM try to expose the more basic controls, but I thought I'd ask. The Multifunction knobs change purpose depending on what tools are selected in the microscope GUI.

My intention is to make a simple GUI / hotkey-controlled approach for helping with doing the basic alignment. Something also the less tech-savvy users could take advantage of. Currently we're using the following control pad simulator, which leaves some to be desired. I've sent Thermo Fisher an email about this today, but I thought I'd check in here as well.

[image]

stefsmeets, 2020-04-22:
Hi @thomasaarholt 
Good question! With `instamatic` this is not possibible. It maps what is available through the API, and  I have not come accross any reference of the multi-function knob (at least for JEOL). I'm not so sure for FEI though, as I don't have access to one.

thomasaarholt, 2020-04-22:
Hehe, that's what I thought. I did not come across mention of it in the TIA scripting handbook either. Hopefully Thermo will get back to me with something. The existence of the app in the screenshot gives me a little bit of hope :)

closed 2020-04-23 by thomasaarholt