instamatic-dev/instamatic #33: Debugging with Visual Studio Code

kttn8769, 2020-04-24:
Hi,

I'm interested in this software and trying to use it now,

It is very helpful for me if I can track the process step by step with VSCode python debugger, but so far with no success. (The debugger will not step-in inside TEMController.initialize())

If you know the way to use instamatic with VSCode debugger, could you please share how to do it?

Thanks

stefsmeets, 2020-04-24:
Hi @kttn8769, I'm not a VSCode user myself. Is there anything in particular you are running into?

kttn8769, 2020-04-24:
Hi @stefsmeets, yes, when I executed the following code in simulation mode,
```
from instamatic import TEMController
ctrl = TEMController.initialize()
```
the following error was raised.

> Microscope: simulate (server)
> Protocol: `pickle`
> Starting TEM server (localhost:8088 on pid=21212)
> Exception in thread None:
> Traceback (most recent call last):
>   File "C:\Users\kttn8\AppData\Local\Programs\Python\Python38\lib\threading.py", line 932, in _bootstrap_inner
>     self.run()
>   File "C:\Users\kttn8\AppData\Local\Programs\Python\Python38\lib\site-packages\instamatic-1.3.0-py3.8.egg\instamatic\server\tem_server.py", line 46, in run
>     self.tem = Microscope(name=self._name, use_server=False)
>   File "C:\Users\kttn8\AppData\Local\Programs\Python\Python38\lib\site-packages\instamatic-1.3.0-py3.8.egg\instamatic\TEMController\microscope.py", line 55, in Microscope
>     cls = get_tem(name)
>   File "C:\Users\kttn8\AppData\Local\Programs\Python\Python38\lib\site-packages\instamatic-1.3.0-py3.8.egg\instamatic\TEMController\microscope.py", line 27, in get_tem
>     raise ValueError(f'No such microscope interface: `{interface}`')
> ValueError: No such microscope interface: `None`
> Connected to TEM server (localhost:8088)
> Camera    : simulate (stream)

instamatic version: commit 224eb90 (master HEAD)
python version: 3.8.2
instamatic config: as generated by default

I opened this issue to see how this happens...

stefsmeets, 2020-04-27:
Yep, that shouldn't happen. Thanks for reporting, I will have a look at it tomorrow!

stefsmeets, 2020-04-28:
Yeah, I get the same here. I have just pushed the fix. As a work-around you can also set `simulate: True` or `use_tem_server: False` in `settings.yaml`.

kttn8769, 2020-04-28:
Now it's working. Thank you!

closed 2020-04-28 by stefsmeets