instamatic-dev/instamatic #34: ValueError: No such microscope interface

kttn8769, 2020-05-01:
Hi,

I have added the following microscope config as config/microscope/jeol-2100hc.yaml,
```
interface: jeol
ranges:
  diff: [15, 20, 25, 30, 40, 50, 60, 80, 100, 120, 150, 200, 250, 300]
  lowmag: [50, 60, 80, 100, 120, 150, 200, 250, 300, 400, 500, 600, 800, 1000,
    1200, 1500, 2000]
  mag1: [1000, 1200, 1500, 2000, 2500, 3000, 4000, 5000, 6000, 8000, 10000,
    12000, 15000, 20000, 25000, 30000, 40000, 50000, 60000, 80000, 100000,
    120000, 150000, 200000, 250000, 300000, 400000, 500000, 600000, 800000]
  mag2: [1000, 1200, 1500, 2000, 2500, 3000, 4000, 5000, 6000, 8000, 10000,
    12000, 15000, 20000, 25000, 30000, 40000, 50000, 60000, 80000, 100000,
    120000, 150000, 200000, 250000, 300000, 400000, 500000, 600000, 800000]
  samag: [5000, 6000, 8000, 10000, 12000, 15000, 20000, 25000, 30000, 40000,
    50000, 60000, 80000, 100000, 120000, 150000, 200000, 250000, 300000,
    400000]
wavelength: 0.025079
```

And edited config/settings.yaml such that
```
microscope: jeol-2100hc
camera: simulate
calibration: simulate
```

But the following error was raised.
```
例外が発生しました: ValueError
No such microscope interface: `jeol-2100hc`
  File "C:\Users\JEOL\Softwares\instamatic\instamatic\TEMController\microscope.py", line 27, in get_tem
    raise ValueError(f'No such microscope interface: `{interface}`')
  File "C:\Users\JEOL\Softwares\instamatic\instamatic\TEMController\microscope_client.py", line 118, in _init_dict
    tem = get_tem(self.name)
  File "C:\Users\JEOL\Softwares\instamatic\instamatic\TEMController\microscope_client.py", line 70, in __init__
    self._init_dict()
  File "C:\Users\JEOL\Softwares\instamatic\instamatic\TEMController\microscope.py", line 53, in Microscope
    tem = MicroscopeClient(name=name)
  File "C:\Users\JEOL\Softwares\instamatic\instamatic\TEMController\TEMController.py", line 49, in initialize
    tem = Microscope(tem_name, use_server=use_tem_server)
  File "C:\Users\JEOL\Softwares\instamatic\instamatic\main.py", line 147, in main
    ctrl = TEMController.initialize(stream=True)
  File "C:\Users\JEOL\Softwares\instamatic\instamatic\main.py", line 163, in <module>
    main()
```

I think the variable "interface" should have a value of "jeol".
Am I doing something wrong? 

Thanks

stefsmeets, 2020-05-01:
Thanks for reporting, I get this too. I worked on this recently, but clearly I didn't iron out all the bugs yet. Try calling the file `jeol.yaml` and putting `microscope: jeol` as a work-around. I will have a look at it today.

kttn8769, 2020-05-01:
I'll follow the work-around, thank you!

stefsmeets, 2020-05-01:
I found the issue in the mean time, I will upload a fix in a bit. It is related to #33

commit 01c93b41 (2020-05-01): Fix #34, ValueError: No such microscope interface
closed 2020-05-01 by stefsmeets