instamatic-dev/instamatic #36: Definition of camera.camera_client.CamClient.get_attrs()

kttn8769, 2020-05-19:
Hi,

I had a question while I was reading the CamClient class code.

CamClient.init_attr_dict() method is defined as follows,

https://github.com/stefsmeets/instamatic/blob/2be8cdb1e2337b1df437868a8de50878d178bd88/instamatic/camera/camera_client.py#L143-L145

I am curious where get_attrs() method is defined for CamClient class.

stefsmeets, 2020-05-19:
Hi @kttn8769,

It's defined here: https://github.com/stefsmeets/instamatic/blob/2be8cdb1e2337b1df437868a8de50878d178bd88/instamatic/server/cam_server.py#L130

It's used for my own implementation of RPC. It's a bit of a hack to enable tab completion IPython when the camera (or tem for that matter) are run in host/server mode.

kttn8769, 2020-05-19:
Thank you @stefsmeets ,

then, CamServer.get_attrs is somehow incorporated into CamClient class??
It is difficult for me to see such interaction in the code...

Sorry if this is a silly question. Maybe I'm overlooking or misunderstand something.

stefsmeets, 2020-05-19:
Yeah, it mimicks the attributes from the actual camera interface in CamClient, because it goes over a socket connection to enable interprocess communication. Therefore CamClient does not know what it is looking at, it just passes the attributes/function calls to the actual interface (e.g. https://github.com/stefsmeets/instamatic/blob/2be8cdb1e2337b1df437868a8de50878d178bd88/instamatic/camera/camera_simu.py#L11). The `.get_attrs` method simply copies the function signatures across so that tab completion and doc strings are available in IPython.

kttn8769, 2020-05-19:
Ok, I understand the point.

Thank you for your kind explanation!

closed 2020-05-19 by stefsmeets