instamatic-dev/instamatic #37: AttributeError: 'ConfigObject' object has no attribute 'pixelsize_diff'

kttn8769, 2020-06-01:
Hi,

I have noticed that config.calibration.pixelsize_diff is not accessible after importing instamatic.config.

Accessing `config.calibration.pixelsize_diff` throws an exception `AttributeError: 'ConfigObject' object has no attribute 'pixelsize_diff'`.

Is this an expected behavior?
I'm using commit fa6af761d842b2300ca86fef5f0431af3a2e3ebb in the sumulation mode.

And could you please show me in which code the attribute pixelsize_diff is defined?

Sorry I'm asking you many times these days.
Thanks!

stefsmeets, 2020-06-02:
Hi @kttn8769, try: `config.calibration['diff']['pixelsize']`

stefsmeets, 2020-06-02:
> And could you please show me in which code the attribute pixelsize_diff is defined?

The values are defined in your calibration yaml: `instamatic -l calibration` in the config. In the code it's here: https://github.com/stefsmeets/instamatic/blob/master/instamatic/config/__init__.py#L96

Can you tell me where you found the syntax you used? I changed this already a while ago, because accessing the config in this way caused problems, but I noticed some references in the code that I forgot to update.

kttn8769, 2020-06-02:
Hi @stefsmeets ,

Ok, I understand the syntax is old-style, thank you.

I noticed this while I was reading instamatic.experiments.cred.experiment.Experiment.start_collection method to see how cRED is implemented.

By the way, I found two possible typos,
https://github.com/stefsmeets/instamatic/blob/fbb001920cfbc3ecc1f6026e2f74433232118729/instamatic/main.py#L126
https://github.com/stefsmeets/instamatic/blob/fbb001920cfbc3ecc1f6026e2f74433232118729/instamatic/gui/debug_frame.py#L9

config.locations instead of config.location?

stefsmeets, 2020-06-02:
Yep, good point! Thanks for the heads-up. Somehow I missed these variables when sweeping through the code.

I should find a way to incorporate the experimental part in the automated tests to catch this.

closed 2020-06-03 by stefsmeets