instamatic-dev/instamatic #44: UnboundLocalError: local variable 'status' referenced before assignment

rds-itga, 2020-10-16:
Hi everybody,

I've just installed instamatic via pip install instamatic
I tried to run this short code:

from instamatic import TEMController
ctrl = TEMController.initialize()

but I got this error:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\ITGA\AppData\Local\Programs\Python\Python38-32\lib\site-packages\instamatic\TEMController\TEMController.py", line 66, in initialize
    ctrl = _ctrl = TEMController(tem=tem, cam=cam)
  File "C:\Users\ITGA\AppData\Local\Programs\Python\Python38-32\lib\site-packages\instamatic\TEMController\TEMController.py", line 121, in __init__
    print(self)
  File "C:\Users\ITGA\AppData\Local\Programs\Python\Python38-32\lib\site-packages\instamatic\TEMController\TEMController.py", line 125, in __repr__
    return (f'Mode: {self.tem.getFunctionMode()}\n'
  File "C:\Users\ITGA\AppData\Local\Programs\Python\Python38-32\lib\site-packages\instamatic\TEMController\microscope_client.py", line 92, in wrapper
    return self._eval_dct(dct)
  File "C:\Users\ITGA\AppData\Local\Programs\Python\Python38-32\lib\site-packages\instamatic\TEMController\microscope_client.py", line 106, in _eval_dct
    if status == 200:
UnboundLocalError: local variable 'status' referenced before assignment

Could you help me to solve it.
Thank you
Regards

stefsmeets, 2020-10-19:
Hi @rds-itga , yup, this seems like a bug. Thanks for reporting. Should not be hard to fix, I will try to have a look at it today or tomorrow.

stefsmeets, 2020-10-19:
Hi @rds-itga , this is a bit of an odd error, meaning that it should not happen. That means something else is not working correctly. Can you tell me a little bit more about your environment and did you do anything else before running the command? Did you change the config? Can you print the full output? Thanks!

rds-itga, 2020-10-21:
Hi,
here are all the steps I followed:
```

pip install instamatic
Collecting instamatic
  Downloading instamatic-1.6.0-py3-none-any.whl (3.1 MB)
     |████████████████████████████████| 3.1 MB 2.2 MB/s
Collecting pyserialem>=0.3.0
  Downloading pyserialem-0.3.1-py3-none-any.whl (36 kB)
Collecting pywinauto>=0.6.8
  Downloading pywinauto-0.6.8-py2.py3-none-any.whl (362 kB)
     |████████████████████████████████| 362 kB 3.3 MB/s
Requirement already satisfied: scipy>=1.3.2 in c:\users\rds\appdata\local\continuum\anaconda3\envs\openclassroom\lib\site-packages (from instamatic) (1.5.0)
Collecting lmfit>=1.0.0
  Downloading lmfit-1.0.1.tar.gz (258 kB)
     |████████████████████████████████| 258 kB 3.3 MB/s
Requirement already satisfied: pandas>=1.0.0 in c:\users\rds\appdata\local\continuum\anaconda3\envs\openclassroom\lib\site-packages (from instamatic) (1.0.5)
Collecting comtypes>=1.1.7
  Downloading comtypes-1.1.7.zip (180 kB)
     |████████████████████████████████| 180 kB 3.2 MB/s
Collecting mrcfile>=1.1.2
  Downloading mrcfile-1.1.2-py2.py3-none-any.whl (41 kB)
     |████████████████████████████████| 41 kB 92 kB/s
Requirement already satisfied: pyyaml>=5.3 in c:\users\rds\appdata\local\continuum\anaconda3\envs\openclassroom\lib\site-packages (from instamatic) (5.3.1)
Collecting virtualbox>=2.0.0
  Downloading virtualbox-2.0.0-py2.py3-none-any.whl (250 kB)
     |████████████████████████████████| 250 kB 3.3 MB/s
Requirement already satisfied: matplotlib>=3.1.2 in c:\users\rds\appdata\local\continuum\anaconda3\envs\openclassroom\lib\site-packages (from instamatic) (3.2.2)
Collecting h5py>=2.10.0
  Downloading h5py-2.10.0-cp37-cp37m-win_amd64.whl (2.5 MB)
     |████████████████████████████████| 2.5 MB 1.3 MB/s
Collecting scikit-image>=0.17.1
  Downloading scikit_image-0.17.2-cp37-cp37m-win_amd64.whl (11.5 MB)
     |████████████████████████████████| 11.5 MB 3.3 MB/s
Requirement already satisfied: pillow>=7.0.0 in c:\users\rds\appdata\local\continuum\anaconda3\envs\openclassroom\lib\site-packages (from instamatic) (7.2.0)
Requirement already satisfied: ipython>=7.11.1 in c:\users\rds\appdata\local\continuum\anaconda3\envs\openclassroom\lib\site-packages (from instamatic) (7.13.0)
Collecting tqdm>=4.41.1
  Downloading tqdm-4.50.2-py2.py3-none-any.whl (70 kB)
     |████████████████████████████████| 70 kB 1.4 MB/s
Requirement already satisfied: numpy>=1.17.3 in c:\users\rds\appdata\local\continuum\anaconda3\envs\openclassroom\lib\site-packages (from instamatic) (1.18.1)
Collecting tifffile>=2019.7.26.2
  Downloading tifffile-2020.10.1-py3-none-any.whl (152 kB)
     |████████████████████████████████| 152 kB 6.8 MB/s
Requirement already satisfied: six in c:\users\rds\appdata\local\continuum\anaconda3\envs\openclassroom\lib\site-packages (from pywinauto>=0.6.8->instamatic) (1.14.0)
Requirement already satisfied: pywin32 in c:\users\rds\appdata\local\continuum\anaconda3\envs\openclassroom\lib\site-packages (from pywinauto>=0.6.8->instamatic) (227)
Collecting asteval>=0.9.16
  Downloading asteval-0.9.19.tar.gz (207 kB)
     |████████████████████████████████| 207 kB ...
Collecting uncertainties>=3.0.1
  Downloading uncertainties-3.1.4-py2.py3-none-any.whl (246 kB)
     |████████████████████████████████| 246 kB 3.3 MB/s
Requirement already satisfied: pytz>=2017.2 in c:\users\rds\appdata\local\continuum\anaconda3\envs\openclassroom\lib\site-packages (from pandas>=1.0.0->instamatic) (2019.3)
Requirement already satisfied: python-dateutil>=2.6.1 in c:\users\rds\appdata\local\continuum\anaconda3\envs\openclassroom\lib\site-packages (from pandas>=1.0.0->instamatic) (2.8.1)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in c:\users\rds\appdata\local\continuum\anaconda3\envs\openclassroom\lib\site-packages (from matplotlib>=3.1.2->instamatic) (2.4.6)
Requirement already satisfied: cycler>=0.10 in c:\users\rds\appdata\local\continuum\anaconda3\envs\openclassroom\lib\site-packages (from matplotlib>=3.1.2->instamatic) (0.10.0)
Requirement already satisfied: kiwisolver>=1.0.1 in c:\users\rds\appdata\local\continuum\anaconda3\envs\openclassroom\lib\site-packages (from matplotlib>=3.1.2->instamatic) (1.2.0)
Collecting imageio>=2.3.0
  Downloading imageio-2.9.0-py3-none-any.whl (3.3 MB)
     |████████████████████████████████| 3.3 MB 6.8 MB/s
Collecting networkx>=2.0
  Downloading networkx-2.5-py3-none-any.whl (1.6 MB)
     |████████████████████████████████| 1.6 MB 6.4 MB/s
Collecting PyWavelets>=1.1.1
  Downloading PyWavelets-1.1.1-cp37-cp37m-win_amd64.whl (4.2 MB)
     |████████████████████████████████| 4.2 MB 3.3 MB/s
Requirement already satisfied: pickleshare in c:\users\rds\appdata\local\continuum\anaconda3\envs\openclassroom\lib\site-packages (from ipython>=7.11.1->instamatic) (0.7.5)
Requirement already satisfied: traitlets>=4.2 in c:\users\rds\appdata\local\continuum\anaconda3\envs\openclassroom\lib\site-packages (from ipython>=7.11.1->instamatic) (4.3.3)
Requirement already satisfied: decorator in c:\users\rds\appdata\local\continuum\anaconda3\envs\openclassroom\lib\site-packages (from ipython>=7.11.1->instamatic) (4.4.2)
Requirement already satisfied: jedi>=0.10 in c:\users\rds\appdata\local\continuum\anaconda3\envs\openclassroom\lib\site-packages (from ipython>=7.11.1->instamatic) (0.15.2)
[... 33 lines cut ...]
Python 3.7.7 (default, Mar 23 2020, 23:19:08) [MSC v.1916 64 bit (AMD64)] :: Anaconda, Inc. on win32
Type "help", "copyright", "credits" or "license" for more information.
[... cut ...]
    return (f'Mode: {self.tem.getFunctionMode()}\n'
  File "C:\Users\rds\AppData\Local\Continuum\anaconda3\envs\OpenClassroom\lib\site-packages\instamatic\TEMController\microscope_client.py", line 92, in wrapper
    return self._eval_dct(dct)
  File "C:\Users\rds\AppData\Local\Continuum\anaconda3\envs\OpenClassroom\lib\site-packages\instamatic\TEMController\microscope_client.py", line 106, in _eval_dct
    if status == 200:
UnboundLocalError: local variable 'status' referenced before assignment
>>>
```
Regards

rds-itga, 2020-10-21:
defaults.yaml

```
gridmontage:
  gridspec:
    direction: 'downup'
    zigzag: true
    flip: true
  overlap: 0.1
montage:
  from_serialem_mrc:
    direction: 'downup'
    zigzag: true
    flip: true
    rot90: 3
    flipud: true
    fliplr: false
predicrystal:
  location: 'C:\predicrystal'
  classifier: holey
  filter_distance: 2.0
```
settings.yaml
```
microscope: simulate
camera: simulate
calibration: simulate

# Global toggle to force simulated camera/microscope interface
simulate: False

data_directory: C:/instamatic
#flatfield: C:/instamatic/flatfield.tiff
flatfield:

# Run the TEM connection in a different process (recommended)
use_tem_server: True
tem_server_host: 'localhost'
tem_server_port: 8088
tem_require_admin: False
tem_communication_protocol: 'pickle'  # pickle, json, msgpack, yaml

# Run the Camera connection in a different process
use_cam_server: False
cam_server_host: 'localhost'
cam_server_port: 8087
cam_use_shared_memory: true

# Submit collected data to an indexing server (CRED only)
use_indexing_server_exe: False
indexing_server_exe: 'instamatic.dialsserver.exe'
indexing_server_host: 'localhost'
indexing_server_port: 8089
dials_script: 'E:/cctbx/dials_script.bat'

# JEOL only, automatically set the rotation speed via Goniotool (instamatic.goniotool)
use_goniotool: False
goniotool_server_host: 'localhost'
goniotool_server_port: 8090

# For InsteaDMatic to control the rotation speed on a FEI/TFS system
fei_server_host: '192.168.12.1'
fei_server_port: 8091

# Automatically submit the data to an indexing server running in a VM (VirtualBox)
use_VM_server_exe: False
VM_server_exe: 'instamatic.VMserver.exe'
VM_server_host: 'localhost'
VM_server_port: 8092
VM_ID: "Ubuntu 14.04.3"
VM_USERNAME: "lab6"
VM_PWD: "testtest"
VM_STARTUP_DELAY: 50
VM_DESKTOP_DELAY: 20
VM_SHARED_FOLDER: F:\SharedWithVM

# Testing variables
cred_relax_beam_before_experiment: false
cred_track_stage_positions: false

# Here the panels for the GUI can be turned on/off/reordered
modules:
  - 'cred'
  - 'cred_tvips'
  - 'cred_fei'
  - 'sed'
  - 'autocred'
  - 'red'
  - 'machine_learning'
  - 'ctrl'
  - 'debug'
  - 'about'
  - 'console'
  - 'io'

```

stefsmeets, 2020-10-22:
Good stuff. I can fix the error quite easily, but it probably will not address the underlying issue.

You can try:

```yaml
use_tem_server: False
```

This will avoid the issue altogether by running the interface in the same process (not ideal, but OK for scripting work).

Alternatively, can you try to first run:

```bash
instamatic.temserver
```

And then in a different tab, do the thing from before:

```python
from instamatic import TEMController
ctrl = TEMController.initialize()
```

rds-itga, 2020-10-27:
Hi, with
`use_tem_server: False`

it seems to work, here is what the the terminal returned to me:

```(OpenClassroom) C:\Users\rds>python
Python 3.7.7 (default, Mar 23 2020, 23:19:08) [MSC v.1916 64 bit (AMD64)] :: Anaconda, Inc. on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from instamatic import TEMController
Config directory: C:\Users\rds\AppData\Roaming\instamatic\config
>>>
>>> ctrl = TEMController.initialize()
Microscope: simulate
Camera    : simulate (stream)

Mode: mag1
High tension: 200 kV
Current density: 101870.12 pA/cm2
GunShift(x=3770, y=58589)
GunTilt(x=63225, y=43005)
BeamShift(x=46346, y=40527)
BeamTilt(x=62627, y=64505)
ImageShift1(x=45491, y=29424)
ImageShift2(x=15383, y=27640)
DiffShift(x=44088, y=41496)
Stage(x=32280.0, y=-15346.0, z=148.0, a=6.0, b=-36.0)
Magnification(value=25000, index=10)
DiffFocus(value=n/a)
Brightness(value=12545)
SpotSize(1)
Saved alignments: ('neutral.yaml',)```

Thank you very much for your help and your reactivity. :)

Regards

stefsmeets, 2020-10-28:
No worries!

closed 2020-10-28 by stefsmeets

rds-itga, 2020-11-20:
Hi again,

I just changed default tem_server_port 8088 into 8087 (as i don't use a cam_server) and now it is working in tem_server mode.
I don't know why it does not work with port 8088, maybe another application is using it on my machine but how to know it
Finally here is the config file i used and which is working

``` ruby
microscope: simulate
camera: simulate
calibration: simulate

# Global toggle to force simulated camera/microscope interface
simulate: false

data_directory: C:\instamatic
#flatfield: C:/instamatic/flatfield.tiff
flatfield:

# Run the TEM connection in a different process (recommended)
use_tem_server: true
tem_server_host: 'localhost'
tem_server_port: 8087
tem_require_admin: false
tem_communication_protocol: 'pickle'  # pickle, json, msgpack, yaml

# Run the Camera connection in a different process
use_cam_server: false
cam_server_host: 'localhost'
cam_server_port: 8087
cam_use_shared_memory: true

# Submit collected data to an indexing server (CRED only)
use_indexing_server_exe: false
indexing_server_exe: 'instamatic.dialsserver.exe'
indexing_server_host: 'localhost'
indexing_server_port: 8089
dials_script: 'E:/cctbx/dials_script.bat'

# JEOL only, automatically set the rotation speed via Goniotool (instamatic.goniotool)
use_goniotool: false
goniotool_server_host: 'localhost'
goniotool_server_port: 8090

# For InsteaDMatic to control the rotation speed on a FEI/TFS system
fei_server_host: '192.168.12.1'
fei_server_port: 8091

# Automatically submit the data to an indexing server running in a VM (VirtualBox)
use_VM_server_exe: false
VM_server_exe: 'instamatic.VMserver.exe'
VM_server_host: 'localhost'
VM_server_port: 8092
VM_ID: "Ubuntu 14.04.3"
VM_USERNAME: "lab6"
VM_PWD: "testtest"
VM_STARTUP_DELAY: 50
VM_DESKTOP_DELAY: 20
VM_SHARED_FOLDER: F:\SharedWithVM

# Testing variables
cred_relax_beam_before_experiment: false
cred_track_stage_positions: false

# Here the panels for the GUI can be turned on/off/reordered
modules:
  - 'cred'
  - 'cred_tvips'
  - 'cred_fei'
  - 'sed'
  - 'autocred'
  - 'red'
  - 'machine_learning'
  - 'ctrl'
  - 'debug'
  - 'about'
  - 'console'
  - 'io'
```

Regards

stefsmeets, 2020-11-20:
Hi @rds-itga , nice debugging there! Thanks for letting me know. The choice of port is somewhat abritrary, but it is indeed possible that it was already in use (or maybe some hanging process). `instamatic` will try to connect to a port if it sees that it is open, but it does not check what it connects too 😅