instamatic-dev/instamatic #46: Interface and microscope

rds-itga, 2020-11-05:
Hi,
I have a question, to use the jeol-1400 configuration
i filled the field microscope: jeol-1400 within the settings.yaml config file but do i have to call

`ctrl = TEMController.initialize(tem_name='jeol', cam_name=None)`
or
`ctrl = TEMController.initialize(tem_name='jeol-1400', cam_name=None)`
 
Thanks

stefsmeets, 2020-11-05:
Hi @rds-itga , I think I understand, it also still gets me confused sometimes 😅 What you describe fits with how it is intended to work.

The idea is that if you filled out the `settings.yaml` then you can use:

```python
ctrl = TEMController.initialize()
```

Then it will use the value configured in `settings.yaml`

If you set `tem_name='jeol'`, then it will override the value you set, and use the given value instead. This seems to be what is happening.

If you want to be really in the clear, just have a single file called `jeol.yaml`.

rds-itga, 2020-11-05:
Ok, I understand, it is clear to me now.
Thank you very much!
Regards

closed 2021-05-25 by stefsmeets