instamatic-dev/instamatic #51: Indexing an electron diffraction pattern

Anudha, 2021-06-24:
This [link](https://github.com/instamatic-dev/instamatic/blob/master/docs/programs.md#instamaticbrowser) suggests that instamatic browser is a program for indexing electron diffraction patterns.  

The command is given as this:

instamatic.browser images/*.tiff -r results.csv

Is there an example indexing an electron diffraction or finding the zone axis based on a known crystal structure?

stefsmeets, 2021-06-25:
Hi @Anudha , thanks for pointing this out, it was at some point built-in to this program. I guess I forgot to update this part of the documentation. I split off all the indexing routines into its own package: `https://github.com/stefsmeets/problematic/`. You may want to have a look at that (although there surely are better options by now 😅).

Anudha, 2021-06-28:
Hi Stefs Meets, 

Thanks for the link!  I tried it out and worked through some errors.  I am posting them here.  If you have moved on from this work, no problem! Public posting will keep the conversation going for anyone interested and new users :)  Yes, I am trying out a number of other codes too.  

**No module named 'problematic.get_score_cy’**
Commented out the relevant line to run for an initial pass
**No module named ‘gcd’ in fractions.py**  
Commented out the relevant line to run for an initial pass

Anudha, 2021-06-28:
First output is 3 figures.  
[image]

stefsmeets, 2021-07-01:
Thanks @Anudha , what code is this output from?

No worries if you found other code that works better. If so, then also let me know so we can close this issue. I don't have much time to maintain `problematic` anymore, although what is in the repo should work.