instamatic-dev/instamatic #56: win32 and GonioTool

rds-itga, 2021-11-10:
Hi again,
sorry for disturbing but I am facing a new issue I cannot solve, I tried to use goniotool.py but I am stuck at the importing step
```
from pywinauto import Application
```
I get this error:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\RDS\AppData\Local\Programs\Python\Python38\lib\site-packages\pywinauto\__init__.py", line 59, in <module>
    import win32api  # noqa: E402
ImportError: DLL load failed while importing win32api: %1 n’est pas une application Win32 valide.
```
Here is the python version I am using:
```
Python 3.8.6 (tags/v3.8.6:db45529, Sep 23 2020, 15:52:53) [MSC v.1927 64 bit (AMD64)] on win32
Instamatic 1.6.0
```
If you could help me to solve this problem I would be very thankful!
Regards

stefsmeets, 2021-11-11:
Hi, I don't recommend to use goniotool.py unless you know what you are doing. I did not intend for others to use it, as it is a very hacky way to manipulate the goniometer rotation speed on some JEOL microscopes.

The error you get is as far as I know a bug with win32api, the solution proposed here worked for me in the past:
https://stackoverflow.com/questions/58612306/how-to-fix-importerror-dll-load-failed-while-importing-win32api

rds-itga, 2021-11-16:
Hi,
thank you for your answer, I will try your link.
I was interested in using GonioTool because we use JEOL microscope and we already use GonioTool manually.
You claim that it is faster using GonioTool
```
# faster stage readout using gonio2
# self.gonio2.GetPosition() -> get stage position, 78 ms
# self.stage3.GetPos() -> 277 ms
``` 
So it could be interesting for performances optimization.
Regards

stefsmeets, 2021-11-16:
Ah yes, that's the api to retrieve the stage position. On our microscope (JEOL 2100), using `gonio2` gave slightly faster readout of the stage position only. But from my experience it did not work well on newer microscopes, and `stage3` is more reliable, so better to use that one. Note that `gonio2` and `stage3` are just the names of the modules of the COM api.

`goniotool.py` is unrelated to that. It hacks into the gui of `goniotool.exe` to be able to control it externally. We used it to automatically adjust the rotation speed (via f1/rate->TX, maybe you do the same manually).

rds-itga, 2021-11-16:
Ok I understand, I thought I needed ``goniotool.py`` to use gonio2, my mistake.
First I'll try ``gonio2`` and see what happens (We are using JEOL 1400).
Maybe I will not need ``goniotool.py`` at the end.
Thank you very much for your help.
Regards

stefsmeets, 2021-11-16:
No worries! I will then close this issue (feel free to re-open / open a new issue).

closed 2021-11-16 by stefsmeets