instamatic-dev/instamatic #62: Adding support for MerlinEM Medipix3 detector

magnunor, 2023-03-24:
We're planning on starting to utilize Instamatic for automating parts of our TEM workflows, which includes MerlinEM detectors. So at some point in the future (not sure exactly when) we'll have a pull request which includes remote control of the Quantum Detector MerlinEM detectors via the TCP/IP API.

This will most likely be via incorporating the code from my old library [Merlin Interface](https://gitlab.com/fast_pixelated_detectors/merlin_interface), which I can relicense to work with Instamatic's license.

At first this will most likely be just the control part, without getting data.

stefsmeets, 2023-03-26:
Hi @magnunor great to hear you are interested in contributing some code to instamatic! Merlin support is definitely something that would be great to have. I have been in contact with Quantum Detectors over the last couple of weeks to support the Merlin detector for the data collection part. So I think our efforts will be quite complementary :-)

GearoidM, 2023-05-11:
Hi @stefsmeets, I noticed another branch for the Merlin. We will be getting a Merlin soon, and we would certainly like to test Instamatic on our system. Have you tested the new software with a Merlin detector?

GearoidM, 2023-05-11:
Thanks for the update! We'll be getting a Merlin with from QD in a few weeks, so I will test it then! I had a chance to set up Instamatic on our microscope using a simulated camera this week. I have some issues that I will raise in another post, but very excited to get it working with the Merlin.

magnunor, 2023-05-12:
@stefsmeets, I noticed the `merlin` branch. Could you make a pull request with this branch? To make it easier to collaborate on the changes. I think I could include some of the code from https://gitlab.com/fast_pixelated_detectors/merlin_interface directly, after relicensing it to BSD.

stefsmeets, 2023-05-15:
Yeah, I can do that later this week. I may also merge to main directly to have a baseline to work from.

stefsmeets, 2023-05-19:
I just merged preliminary support for the merlin detector in #65 

It's still early stages (see limitations in the PR 😅), but the interface at least works well.

magnunor, 2023-05-25:
Nice! I'll try to have a look at this sometime in June, but not 100% sure. In any case, I'll have some students working on this from the autumn.

stefsmeets, 2023-06-27:
The merlin interface is working and seems to be quite stable. So I think this issue can be closed. Feel free to re-open or make a new issue if you run into any issues with testing.