instamatic-dev/instamatic #69: Connection Issue with Merlin

GearoidM, 2023-05-30:
We recently got a Merlin installed on our system (JEOL 2200). I have installed Instamatic v1.9.0 on both the Merlin PC and microscope PC. The Merlin PC does not have the TEMCOM libraries, and is disconnected from the microscope PC by default. 

**When running Instamatic on the Merlin PC:**

If I choose JEOL as the microscope in the settings.yaml file (i.e., `microscope: jeol, camera: merlin, calibration: simulate`) I get:

```
Config directory: C:\Users\Merlin\AppData\Roaming\instamatic\config
Microscope: jeol (server)
Starting TEM server (localhost:8088 on pid=4600)
Exception in thread None:
Traceback (most recent call last):
  File "C:\Users\Merlin\AppData\Local\Programs\Python\Python38\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "C:\Users\Merlin\AppData\Local\Programs\Python\Python38\lib\site-packages\instamatic-1.9.0-py3.8.egg\instamatic\server\tem_server.py", line 46, in run
    self.tem = Microscope(name=self._name, use_server=False)
  File "C:\Users\Merlin\AppData\Local\Programs\Python\Python38\lib\site-packages\instamatic-1.9.0-py3.8.egg\instamatic\TEMController\microscope.py", line 55, in Microscope
    tem = cls(name=name)
  File "C:\Users\Merlin\AppData\Local\Programs\Python\Python38\lib\site-packages\instamatic-1.9.0-py3.8.egg\instamatic\TEMController\jeol_microscope.py", line 61, in __init__
    temext = comtypes.client.GetModule(('{CE70FCE4-26D9-4BAB-9626-EC88DB7F6A0A}', 3, 0))
  File "C:\Users\Merlin\AppData\Local\Programs\Python\Python38\lib\site-packages\comtypes\client\_generate.py", line 139, in GetModule
    tlib = _load_tlib(tlib)
  File "C:\Users\Merlin\AppData\Local\Programs\Python\Python38\lib\site-packages\comtypes\client\_generate.py", line 178, in _load_tlib
    return LoadRegTypeLib(GUID(libid), *version)
  File "C:\Users\Merlin\AppData\Local\Programs\Python\Python38\lib\site-packages\comtypes\typeinfo.py", line 478, in LoadRegTypeLib
    _oleaut32.LoadRegTypeLib(byref(GUID(guid)), wMajorVerNum, wMinorVerNum, lcid, byref(tlib))
  File "_ctypes/callproc.c", line 935, in GetResult
OSError: [WinError -2147319779] Library not registered
```

I figured this is because I am missing the TEMCOM library on the Merlin PC, so tried to just proceed with a simulated microscope and the Merlin camera (i.e., `microscope: simulate,  camera: merlin,  calibration: simulate`, and I get the following:

```
C:\Users\JEOL>instamatic
Config directory: C:\Users\JEOL\AppData\Roaming\instamatic\config
Microscope: jeol (server)
Starting TEM server (localhost:8088 on pid=3912)
Connected to TEM server (localhost:8088)
Camera    : merlin (stream)
Traceback (most recent call last):
  File "c:\users\jeol\appdata\local\programs\python\python38\lib\site-packages\i
nstamatic\camera\camera_merlin.py", line 179, in establishConnection
    s_cmd.connect((self.host, self.commandport))
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\jeol\appdata\local\programs\python\python38\lib\runpy.py", line
 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\users\jeol\appdata\local\programs\python\python38\lib\runpy.py", line
 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\JEOL\AppData\Local\Programs\Python\Python38\Scripts\instamatic.
exe\__main__.py", line 7, in <module>
  File "c:\users\jeol\appdata\local\programs\python\python38\lib\site-packages\i
nstamatic\main.py", line 138, in main
    ctrl = TEMController.initialize(stream=True)
  File "c:\users\jeol\appdata\local\programs\python\python38\lib\site-packages\i
nstamatic\TEMController\TEMController.py", line 60, in initialize
    cam = Camera(cam_name, as_stream=stream, use_server=use_cam_server)
  File "c:\users\jeol\appdata\local\programs\python\python38\lib\site-packages\i
nstamatic\camera\camera.py", line 67, in Camera
    cam = cam_cls(name=name)
  File "c:\users\jeol\appdata\local\programs\python\python38\lib\site-packages\i
nstamatic\camera\camera_merlin.py", line 54, in __init__
    self.establishConnection()
  File "c:\users\jeol\appdata\local\programs\python\python38\lib\site-packages\i
nstamatic\camera\camera_merlin.py", line 194, in establishConnection
    raise RuntimeError(
RuntimeError: Could not establish command connection to merlin, (Merlin command
port already connected).
SUCCESS: The process with PID 2272 (child process of PID 3912) has been terminat
ed.
SUCCESS: The process with PID 3912 (child process of PID 3252) has been terminat
ed.
```

I couldn't quite understand what was happening with the port. It seems like there is a conflicting process, but I could not identify what that was. The application to control the Merlin was not running.
In relation to the missing TEMCOM library, I have connected the PCs via an Ethernet cable, and have tried running two instances of Instamatic but still no luck. We would appreciate any information on how to establish communication between the JEOL PC and Merlin PC!

stefsmeets, 2023-05-31:
> The application to control the Merlin was not running.

Ah, I think you have to start the Merlin software first. 

The error here is a bit confusing, I guess it says this on all failed connection attempts. I will change it.

---

> In relation to the missing TEMCOM library, I have connected the PCs via an Ethernet cable, and have tried running two instances of Instamatic but still no luck. We would appreciate any information on how to establish communication between the JEOL PC and Merlin PC!

I think you could try asking your JEOL engineer to set you up with TEMCOM on the Merlin PC. 

If you have instamatic on the JEOL PC and run `instamatic.temserver` there, and put the IP of the JEOL PC under `hostname` on the `settings.yaml` of the Merlin PC. This way the two instances of instamatic should be able to communicate.

JEOL PC:
```
# settings.yaml
use_tem_server: True  # ignored
tem_server_host: 1.2.3.4  # IP of JEOL PC
tem_server_port: 8088
```

Merlin PC:
```
# settings.yaml
use_tem_server: True
tem_server_host: 1.2.3.4  # IP of JEOL PC
tem_server_port: 8088  # Must match PORT of JEOL PC
```

GearoidM, 2023-06-05:
Thanks, @stefsmeets. I managed to establish communication between the JEOL PC and Merlin PC with the information provided above. Previously, I used the wrong IP address in the `tem_server_host:` line, but eventually found the right one.

I tested Instamatic with the Merlin software running and not running but still get the same error. The next time I test things, I will adjust the code to output the exact OSError to be more informative. Anyway, I checked the ports being used by the Merlin software when it is running, and they seem to match the ports in `instamatic/config/camera/merlin.yaml`, i.e. 6341 and 6342 ( and also 6343, which is not shown in this file). I am wondering if perhaps I need to change the IP address from 10.0.0.130 to 127.0.0.1, since I am running Instamatic on the Merlin PC (and instamatic.temserver on the JEOL PC).

GearoidM, 2023-06-05:
I got Instamatic working by changing the IP address in `merlin.yaml` from 10.0.0.130 to 127.0.0.1. Will test tomorrow again to double check, and then close the thread!

stefsmeets, 2023-06-06:
Ah yes, that's a good point. I will change the default to `127.0.0.1` instead of the url I use for testing. I'm planning to write some documentation for how to setup the Merlin detector with instamatic soon.

GearoidM, 2023-06-06:
Yes, I noticed that there is a new version! Works great so far!

closed 2023-06-06 by GearoidM