instamatic-dev/instamatic #95: Write the data acquisition parameters to a more easily parsed file

iverks, 2024-10-22:
Hi, thanks for the great library. 

It would be even better if the data acquisition parameters (e.g. cRED_log.txt) were written to a standard file format (.json, .yaml or .toml) to make it easier to create automation scripts without writing a custom parser.

stefsmeets, 2024-10-22:
Fully agree, I think yaml would be a good choice here. Is this something you would be willing to help out with?

iverks, 2024-10-23:
Yes of course. Do you want it to be as well as or instead of the cRED_log.txt?

stefsmeets, 2024-10-23:
I don't think it hurts to keep it for a while, but I also think yaml is the way forward for this. It should be easy to deprecate the cred_log.txt in a while.

iverks, 2025-06-07:
My work took a different direction than i expected, so I couldn't find time to work on this unfortunately. I suppose it shouldn't be too hard, but it is work nevertheless. Closing this issue, you can reopen if you have the surplus to implement it yourself, but it's not needed by us for now at least.

```py
import yaml

with open("filename.yml", "w") as write_file:
    yaml.dump({
        "parameter": classy.value,
    }, write_file)
```