instamatic-dev/instamatic #104: Expand scope of camera simulation

viljarjf, 2024-11-21:
Currently, the camera simulation is simply an API simulation, with the image being random noise.
This is limiting, as e.g. the experiments cannot be tested properly with this; they require spatially defined crytstals capable of diffraction.

I started working on this in https://github.com/viljarjf/instamatic/tree/camera-simulation.
There, I added a stage and a random crystal structure, randomly dispersed and oriented on a grid.
I plan to support stage rotation, tilting, correct handling of beam shift/tilt ect. later. However, I am not entirely sure how all the microscope parameters affect the images (real and reciprocal), so to begin with, I plan to keep it simple and allow expansions at a later stage.
Currently, the simulation supports a single crystal phase, random dispersion, any number of circular crystals of any size, and a copper grid. Tilting is mostly implemented, but not tested, so it will display a warning for now. 

Expanding this to support more phases, tilting, magnification-induced image rotation and so on should not be too difficult programming-wise, however I'm not too confident that I'll do this correctly without input. Therefore, I'm opening this issue to hopefully have a discussion about what and how things should be simulated.

An improved camera simulation should dramatically expand the features of instamatic that can be properly tested. We can test the experiments, the GUI, calibration (if we implement an accurate simulation). Additionally, it should make further developments much easier, as testing algorithms would not strictly require access to a microscope (at least for initial development).

As for downsides, expanding the scope can easily get carried away. It will necessarily add more dependencies (as is already the case with my branch), which will only get worse with increased accuracy (hopefully! Complex imaging/diffraction simulations **should** be handled by external libraries). Handling inaccurate alignment, astigmatism ect is a good idea, but the added work of implementing this might not be worth it compared to how basic it is to compensate for by the operator.

viljarjf, 2024-11-21:
Timeline-wise, the simple initial implementation is pretty much ready to go. I would like to add some more tests, and try it out some more, but I don't think expanding the initial scope is necessary. As stated, it should be fairly simple to expand, but the scope can be kept simple to start with.

stefsmeets, 2024-11-26:
This is great! Being able to set up a simulation to play with instamatic without installing it on the microscope would be very useful.

I can see the dependencies getting out of scope easily. If you are worried about dependency creep, you could place the simulation dependencies as an 'extra', so you would do `pip install instamitic[simulation]` to install the simulation dependencies (or `instamatic[digital-twin]` if you want to get on board of that hype train).

Another option would be to develop this in a separate module (or possibly package outside of instamatic), and have a separate interface under microscope and camera to interface with this module. I think it makes sense to have this code as separate as possible from anything else.

If you are going this direction, I think the code should be under `instamatic/simulation`, rather than under camera because it will maintain the state of both the microscope and camera, right? In the camera module you could have a very shallow wrapper for interfacing with the simulation code. I also foresee that this will also make the code easier to extend and re-use for others.

Configuration would also be something to think about. How does this initialize a simulated TEM? Can I import the simulation into a notebook? As a user, can I define crystal positions manually? This would be useful for testing.

This would be cool to have:

```python
from instamatic.simulation import DigiTEM

sim = DigiTEM(...)
# alternative
# sim = DigiTEM.from_config('path/to/config.yaml')
sim.add_crystal(x=123, y=321, shape='cube', thickness=10)
sim.add_crystal(...)
...

from instamatic.controller import TEMController

ctrl = TEMController(tem=sim.tem, cam=sim.cam)
ctrl.stage.set_xy(x=123, y=321)
ctrl.get_image()
```