instamatic-dev/instamatic #113: Faulty defocus compensation due to slow first images

iverks, 2025-03-06:
In our setup, for whatever reason the first two images take a long time to acquire, generally 1200 and 1000 ms each.

When the first few defocus frames arrive, the calculation `acquisition_time = (t_start - t0) / (i - 1)` from [line 301 in cred/experiment.py](https://github.com/instamatic-dev/instamatic/blob/20f82851ebad502cf374c359ccadba851905aa98/src/instamatic/experiments/cred/experiment.py#L301) overestimates the real acquisition time and does not properly skip a frame until sufficiently many frames have been taken to reduce the average time. 

One way to solve it could be to use a more robust estimator, like the median frame time instead of the mean.

## Example

Acquisition times. Here you can see that the first two frames take a long time to acquire for whatever reason.
[image]

Time between captures. Here you can see that even though the defocus frames 10 and 20 take a long time, the frames 11 and 21 are not skipped.
[image]

Baharis, 2025-03-06:
Hi Iver, I don't quite have experience with the CRED routine yet but I recently messed with the Instamatic back-ends so I am happy to help you with brainstorming. Which detector interface are you using? As a test, you could try adding one or two `_, _ = self.ctrl.get_image(self.exposure, header_keys=None)` in line 297. To my (inexperienced) eye it seems that whatever camera you are using needs a lot of time to configure itself when collecting the first image. This could be confirmed by collecting a few irrelevant frames before starting the experiment. Also, I have not delved into the reasoning behind these intervals but I can help you to test a "median"-based estimator if you wish.

iverks, 2025-03-06:
We are using the merlin camera interface. I guess I should note that this is no longer a major issue since we recalculate the angles from the timestamps on the tiff images and rewrite the pets.pts config file with the recalculated angles. It is a bit of a footgun nevertheless.

I agree that the time is due to some configuration, as we see the same lag if we change the exposure time. Taking dummy images would probably fix that part of the issue.

Baharis, 2025-03-06:
Based on what I can see in the `CameraMerlin` class, if you want to get rid of unexpected delays when changing the exposure, instead of collecting a dummy image you should just need to call `self.setup_soft_trigger(exposure=exposure)`, with the new value of exposure. It includes a bunch of `merlin_set` commands that would sum up to a second if every one ran 100-200 ms ([link](https://github.com/instamatic-dev/instamatic/blob/20f82851ebad502cf374c359ccadba851905aa98/src/instamatic/camera/camera_merlin.py#L175-L206)). It is not a general solution for the code as different detectors do not need (and thus do not have) this method but if you want to have control over the delay in your personal code, it might be useful.

stefsmeets, 2025-03-10:
I also want to chime in that I haven't actually tested the Merlin code with cRED specifically when I implemented this. I agree with @Baharis analysis that the cRED routine might need a specific warm up step for Merlin specifically.

iverks, 2025-03-21:
> Based on what I can see in the `CameraMerlin` class, if you want to get rid of unexpected delays when changing the exposure, instead of collecting a dummy image you should just need to call [...]

Note: this is not the most recent instamatic, i think it's 2.0.1.  
From our logs:

```log
2024-11-18 11:33:37,207 | camera_merlin:242 | INFO | Change exposure to 0.01 s
2024-11-18 11:33:37,208 | camera_merlin:68 | DEBUG | MPX,0000000020,CMD,STOPACQUISITION
2024-11-18 11:33:37,209 | camera_merlin:176 | DEBUG | MPX,0000000022,CMD,STOPACQUISITION,0
2024-11-18 11:33:37,209 | camera_merlin:68 | DEBUG | MPX,0000000025,SET,ACQUISITIONTIME,10.0
2024-11-18 11:33:38,208 | camera_merlin:133 | DEBUG | MPX,0000000022,SET,ACQUISITIONTIME,0
2024-11-18 11:33:38,208 | camera_merlin:139 | DEBUG | Remembering state for ACQUISITIONTIME value 10.0
2024-11-18 11:33:38,208 | camera_merlin:68 | DEBUG | MPX,0000000027,SET,ACQUISITIONPERIOD,10.0
2024-11-18 11:33:38,219 | camera_merlin:133 | DEBUG | MPX,0000000024,SET,ACQUISITIONPERIOD,0
2024-11-18 11:33:38,219 | camera_merlin:139 | DEBUG | Remembering state for ACQUISITIONPERIOD value 10.0
2024-11-18 11:33:38,219 | camera_merlin:68 | DEBUG | MPX,0000000021,CMD,STARTACQUISITION
2024-11-18 11:33:38,220 | camera_merlin:176 | DEBUG | MPX,0000000023,CMD,STARTACQUISITION,0
2024-11-18 11:33:38,222 | camera_merlin:115 | INFO | Received 14 bytes in 1 steps (0.002117 s)
2024-11-18 11:33:38,223 | camera_merlin:115 | INFO | Received 2049 bytes in 1 steps (0.000040 s)
2024-11-18 11:33:38,223 | camera_merlin:68 | DEBUG | MPX,0000000016,CMD,SOFTTRIGGER
2024-11-18 11:33:38,224 | camera_merlin:176 | DEBUG | MPX,0000000018,CMD,SOFTTRIGGER,0
2024-11-18 11:33:38,368 | camera_merlin:115 | INFO | Received 14 bytes in 1 steps (0.143869 s)
2024-11-18 11:33:38,368 | camera_merlin:255 | INFO | Received header: 1049345 (bytearray(b'MPX,0001049345'))
2024-11-18 11:33:38,378 | camera_merlin:115 | INFO | Received 1049345 bytes in 16 steps (0.009817 s) #First image
2024-11-18 11:33:38,385 | camera_merlin:68 | DEBUG | MPX,0000000016,CMD,SOFTTRIGGER
2024-11-18 11:33:38,386 | camera_merlin:176 | DEBUG | MPX,0000000018,CMD,SOFTTRIGGER,0
2024-11-18 11:33:39,379 | camera_merlin:115 | INFO | Received 1049359 bytes in 26 steps (0.991288 s) #Second image
2024-11-18 11:33:39,385 | camera_merlin:68 | DEBUG | MPX,0000000016,CMD,SOFTTRIGGER
2024-11-18 11:33:39,386 | camera_merlin:176 | DEBUG | MPX,0000000018,CMD,SOFTTRIGGER,0
2024-11-18 11:33:39,407 | camera_merlin:115 | INFO | Received 1049359 bytes in 34 steps (0.021347 s) #Third image
``` 

In addition to the first image taking 1.2s, probably due to the reason you listed, we can see that the second image actually takes a full second to recieve, at which point nothing else is happening (in instamatic at least). 

Shouldn't the fix be before we start rotation rather than after? So line 292 rather than 297? Otherwise we inflate the total rotation angle.

stefsmeets, 2025-03-22:
Yeah, these 2 are the important lines: https://github.com/instamatic-dev/instamatic/blob/20f82851ebad502cf374c359ccadba851905aa98/src/instamatic/experiments/cred/experiment.py#L291-L292 

Once you start tilting the stage and data collection starts, the frame times should be stable. Basically, after `camera.block` is called, instamatic expects stable frame times. So the fix should be before those lines.

Baharis, 2025-12-09:
@iverks Could you mention if you managed to get this issue fixed at your setup and, if so, by what exact means? I would like resolve the issue, either by revisiting it or introducing a patch.

Baharis, 2025-12-15:
> Instead I used the timing metadata from the images to generate a new PETS config file with better angular estimates for each image, which helped a lot.

@iverks This is a very valuable observation! I needed a reminder that `instamatic/processing/ImgConversion.py` simply assumes that `phi = self.start_angle + self.osc_angle * (i - 1)` while both `ImageGetTimeStart` and `ImageGetTimeEnd` are MANDAROTORY header items for every image – which means estimating alpha angle this way directly in Instamatic is definitely doable, although maybe not as a default. Thank you very much for taking your time to respond, it is a bit dissapointing that reproducing the issue may be problematic but if I get to look at the conversion, I'll definitely see if I'm able to enable this method of angle determination.