instamatic-dev/instamatic #115: Latest release of instamatic doesn't work with latest version of tifffile

iverks, 2025-03-21:
The latest release of instamatic doesn't work with latest version of tifffile.
This is already fixed in #102, but it is not released yet. It would be useful with a new release to avoid having to downgrade tifffile.

Additionally i suggest pinning the required tifffile version since it doesn't follow semver, to avoid issues like this in the future.

stefsmeets, 2025-03-22:
Thanks for pointing this out, I made a new release but the release pipeline broke: https://github.com/instamatic-dev/instamatic/actions/runs/14007491363/job/39223209482

stefsmeets, 2025-03-22:
Version 2.1.0 is available on pypi. I'd rather not pin the tifffile version, but let me know if the problem persists.

closed 2025-03-22 by stefsmeets