instamatic-dev/instamatic #118: Make get_movie a generator function to get new images life

Baharis, 2025-03-26:
While reworking the `get_movie` methods in PR #111, and discussing removal of `VideoStream.continuous_collection` in PR #116 , I had an interesting realization. Currently, the `get_movie` "pipeline" is basically not used anywhere in Instamatic. The method is defined for 4 cameras (Base, Simu, Merlin, Serval), two `VideoStream`s (Fake and Live) and the `Controller`. Then, despite having a clear benefit of being gapless (or almost gapless), it is used only in 2 doc-strings, 2 tests, and `VideoStream` / `Controller`. No instamatic `Frame` nor any instamatic `Experiment` uses this method, as evidenced by the fact that before #111 using it was likely to raise error due to concurrency issues, yet it went unnoticed.

One issue with `get_movie` is the fact that it blocks the thread and does not yield any results until the movie is complete. This appears to stand against the design principle of the GUI, where the current image is always dynamically displayed in the `VideoStreamFrame`. This is also an issue for myself, as I am currently working on a general implementation of a cRED for instamatic (tvips is very specific and does not control camera, auto-cRED is much more and uses `get_image`). I want to have a frame that allows running a normal cRED on any setup and displays the movie as it is collected.

The unexpected benefit of `VideoStream.continuous_collection` removed in PR #116 was that it could be used as a boot-leg gap-prone budget version of `get_movie` that does display the images as they are gathered. While I don't think it shows images by default (despite claiming to do so), by utilizing a smart callback function akin ~`lambda img: video_stream.frame = img`, with some changes it could have been potentially used to set the current frame to the image that was just collected:
```python
def continuous_collection(self, exposure=0.1, n=100, callback=None):
    buffer = []
    go_on = True
    i = 0
    self.block()
    while go_on:
        i += 1
        img = self.get_image(exposure=exposure)
        if callback:
            go_on = callback(img)
        else:
            buffer.append(img)
            go_on = i < n    # would require a better stopping condition
    self.unblock()
    if not callback:
        return buffer        # would require returning even if callback's present, or callback that evaluates to false
```

So here is where a thought came to me: what if we modify the implementation of `get_movie`, so that instead of returning a `List[np.ndarray]` at the end of the collection, it becomes a `Generator[np.ndarray, None, None]` method? Currently, all `SomeCamera.get_movie` methods start the data collection, wait for every image, put it into a list, and return the list once it is complete. By replacing the `list.append(image)` with `yield image` and splitting get_movie into two threads, we could have `get_movie` that e.g. immediately produces images whenever they are collected, something like:

[image]

```python
class CameraSimu(CameraBase):
    def get_movie(self, n_frames: int, *, exposure: Optional[float] = None, binsize: Optional[int] = None, **kwargs,) -> Generator[np.ndarray, None, None]:
        for _ in range(n_frames):
            yield self.get_image(exposure=exposure, binsize=binsize)

...

class LiveVideoStream(VideoStream):
    def get_movie(self, n_frames: int, exposure=None, binsize=None):
        request = MovieRequest(n_frames=n_frames, exposure=exposure, binsize=binsize)
        for image in iter(self._get_media(request), None):  # generator that stops when it receives None
            self.frame = image
            images.append(image)
        return image
```

Here I base my example on `CameraSimu` and offer only a crude pseudo-code implementation for `LiveVideoStream`, but I took a look and I believe that both `CameraServal` and `CameraMerlin.get_movie` should be relatively easy to adapt to this pattern. I have not started work on the exact implementation but I would be really interested to hear whether anyone shares my sentiment here.

stefsmeets, 2025-03-27:
Interesting idea, I used something like the proposed pattern here that releases the block by using futures:

https://github.com/instamatic-dev/instamatic/blob/a90dac4f06ca1ea444460e7edfbe6a44016168b6/src/instamatic/controller.py#L541-L563

The idea is quite simple:

```python
future = ctrl.get_future_image()  # non-blocking
(do stuff)
img = future.result()  # blocks until image is ready
```

The trick to get gapless data collection is to ensure that (in your example), the 'Get frame' thread is called when the next frame needs to be called. This means that the `Set frame` (or any other) thread is not blocking during this moment.

Baharis, 2025-03-27:
My original idea was to experiment with `asyncio` and maybe make `get movie` an asynchronous generator function by using `async def get_movie`, but then this must be done carefully because `tkinter` and asyncio apparently really hate each other. The added complexity here comes from the fact that `VideoStream.get_image` needs images ASAP to show them, but `controller.get_image` needs images ASAP to enhance them with metadata, preferably also collected in parallel.

@stefsmeets Can I take your "Interesting idea" as a declaration of "this might be something worth merging when ready"?

stefsmeets, 2025-03-27:
> [@stefsmeets](https://github.com/stefsmeets) Can I take your "Interesting idea" as a declaration of "this might be something worth merging when ready"?

Haha, if you can make it work, then absolutely! The multi-threading bit is difficult to get right, especially because it can act in unexpected ways. I expect it will be a challenge to get the timings right, but this is definitely along the lines of my later work on instamatic (the future image being an example).

Baharis, 2025-03-28:
@stefsmeets That's naturally still WIP but I managed to rewrite `CameraSimu.get_movie`, `VideoStream.get_movie` and `TEMController.get_movie` to be all sequential generators. I was a bit worried about overhead since first `CameraSimu` yields to `VideoStream`, then `VideoStream` yields to `TEMController`, and then `TEMController` yields to the original called but the following code:
```python
print(time.perf_counter())
for i, h in ctrl.get_movie(1000, exposure=0.1): pass
print(time.perf_counter())
```
takes 102.24084 seconds, meaning there was a 2.24 ms overhead for each frame collected. The physical overhead i.e. pixel dead time on our detector is 1 to 2 ms, so being in this range makes me very happy! Nevertheless this might lead to some slight offsets appearing for very long rotations with very short expositions, so I plan to write a calibration class that would take this into account and lower the actual exposition to requested exposition - overhead.

If you are curious about implementation, feel free to check out the following but the weekend calls!
https://github.com/instamatic-dev/instamatic/compare/main...Baharis:instamatic:get_movie_generator