instamatic-dev/instamatic #119: Misformed yaml header due to beam center

iverks, 2025-03-27:
For some reason, the beam_centers are serialized as numpy floats, which causes errors when trying to read the header using yaml.load

## Current version

```yaml
ImageCameraDimensions: [512, 512]
ImageCameraName: merlin
ImageComment: ''
ImageExposureTime: 0.1
ImageGetTime: 1742912131.8822157
ImageGetTimeEnd: 941559.6406307
ImageGetTimeStart: 941558.0765799
ImageResolution: [512, 512]
beam_center: [!!float 'np.float64(291.68)', !!float 'np.float64(284.58)']
```

## Previous version

```yaml
ImageCameraDimensions: [512, 512]
ImageCameraName: merlin
ImageComment: ''
ImageExposureTime: 0.025
ImageGetTime: 1731915693.344416
ImageGetTimeEnd: 257.984739289
ImageGetTimeStart: 257.894768614
ImageResolution: [512, 512]
beam_center: [235.69, 274.28]
```

## Code trying to load metadata

```py
import tifffile as tf
import yaml

 with tf.TiffFile(image) as img:
        page: tf.TiffPage = img.pages[0]
        metadata = yaml.load(page.tags["ImageDescription"].value, Loader=yaml.Loader)
```

## Error message when trying to load 

```
File "./get_aquizition_times.py", line 14, in get_aquizition_time
    metadata = yaml.load(page.tags["ImageDescription"].value, Loader=yaml.Loader)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/site-packages/yaml/__init__.py", line 81, in load
    return loader.get_single_data()
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/site-packages/yaml/constructor.py", line 51, in get_single_data
    return self.construct_document(node)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/site-packages/yaml/constructor.py", line 60, in construct_document
    for dummy in generator:
  File "/site-packages/yaml/constructor.py", line 408, in construct_yaml_seq
    data.extend(self.construct_sequence(node))
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/site-packages/yaml/constructor.py", line 129, in construct_sequence
    return [self.construct_object(child, deep=deep)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/site-packages/yaml/constructor.py", line 129, in <listcomp>
    return [self.construct_object(child, deep=deep)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/site-packages/yaml/constructor.py", line 100, in construct_object
    data = constructor(self, node)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/site-packages/yaml/constructor.py", line 292, in construct_yaml_float
    return sign*float(value)
                ^^^^^^^^^^^^
ValueError: could not convert string to float: 'np.float64(290.13)'
```

Baharis, 2025-03-27:
Seams like an easy fix, on it

stefsmeets, 2025-03-27:
This is in the tiff header, right?

This is the offending line: https://github.com/instamatic-dev/instamatic/blob/e443b832a275555f4ac46824dd0eca78532ac73f/src/instamatic/processing/ImgConversion.py#L272

Calling `find_beam_center`, which returns a numpy array.

https://github.com/instamatic-dev/instamatic/blob/e443b832a275555f4ac46824dd0eca78532ac73f/src/instamatic/tools.py#L115-L132

It's probably best fixed by doing: `h['beam_center'] = float(cx), float(cy)`

Baharis, 2025-03-27:
@stefsmeets @iverks I committed exactly this suggested fix straight to main, so the issue should be gone. There might be problems elsewhere since I noticed that the beam center functions notoriously promise to return `float`, but return `np.float64`.

stefsmeets, 2025-03-27:
@Baharis Thanks! As a matter of procedure, I would prefer such changes to go through a PR so that we can keep track of the fixes in the changelog and so it can be linked to this issue.

Closed by a90dac4f06ca1ea444460e7edfbe6a44016168b6

> There might be problems elsewhere since I noticed that the beam center functions notoriously promise to return `float`, but return `np.float64`

Yeah, that's why I wonder where this bug came from, considering this code has not changed recently (meaning in the last 7 years 😅). I think it is correct for the beam center functions to return a numpy array (the types are a bit off there). But since this goes into the header, I think the change is fairly safe.

closed 2025-03-27 by stefsmeets

iverks, 2025-03-27:
I would guess the bug is from either numpy or pyyaml being updated to a newer version that changes the serialization