instamatic-dev/instamatic #120: 24 bit data stored as 16 bit for Merlin detector

iverks, 2025-04-01:
When we set 24bit mode in config/camera/merlin.yaml our data is stored as 16 bit. It seems like the data is at one point 24 bit but then truncated, wrapping around to becoming small values again. 

Example tiff: https://filesender.sikt.no/?s=download&token=aae593a6-2a36-484f-828c-446cf03716a5
Detector: Merlin 4r

iverks, 2025-04-01:
We are using cRED. The line you point to is probably correct. We use PETS, I wasn't aware it could only handle 16 bit tiffs.
Quick testing shows that rather than clamping, this wraps the integers around. Isn't clamping the integers to `2**16-1` a better solution?

```py
arr = np.array([2**16 + 2] * 10, dtype=np.uint32)
print(arr)
print(arr.astype(np.uint16))
print(
    np.minimum(arr, 2**16 - 1).astype(
        np.uint16,
    )
)
```

```console
[65538 65538 65538 65538 65538 65538 65538 65538 65538 65538]
[2 2 2 2 2 2 2 2 2 2]
[65535 65535 65535 65535 65535 65535 65535 65535 65535 65535]
```

danielnrainer, 2025-04-01:
The "new" PETS 2.0 can handle 16 and 32bit afaik. I just tried a 32bit dataset and it looks fine. Maybe it can also handle 24bit, but this I don't know, maybe best to test and/or just ask Lukáš 😊.

Baharis, 2025-04-01:
@iverks AFAIK PETS 2 should handle 32-bit files correctly. I can ask on our developer discord but I am fairly sure that is the case. I take the image you sent comes straight from the camera? In this case PETS wouldn't have anything to do with it. Or is it from PETS? Would you be able to confirm at which point in the code does the maximum fall from above to below 65535 units? You can add a bunch of `log.warning(image.max())` statements around the code and take a look at the log.

As for the `2**16-1` question, rounding is is not only computationally expensive but also misleading. Would you prefer to occasionally have well-obscured wrong data, or occasionally have clearly wrong data and be able to inquire about it?

iverks, 2025-05-20:
> Which mode are you using for data collection?
> 
> Could be this bit here, one of the goals was to make sure that the data are saved in a way that it could be directly used with data processing codes. This may need to be revised: https://github.com/instamatic-dev/instamatic/blob/main/src/instamatic/processing/ImgConversion.py#L453

We patched our local installation of Instamatic to not convert to 16bit. It works perfectly, I recommend revising. Converting to 16bit can be done in preprocessing anyways.

stefsmeets, 2025-05-20:
Nice to hear! Can you share your code or make a pull request so that we can include it?

iverks, 2025-05-20:
Made a quick PR now, it is however only tested for cRED on a merlin camera. As I wrote, I'm not sure if removing the round is safe

PR #127 (2025-05-20): Remove conversion to uint16
Fixes #120. This is how we did it on our microscope. Afaik the input is always uint, so np.round(img, 0) is a noop. If this is not the case, the rounding should be added back in.
closed 2025-05-20 by stefsmeets