instamatic-dev/instamatic #123: Displaying and enhancing existing frames in the GUI

Baharis, 2025-05-05:
Recently, I have been working on a custom Instamatic branch with a modified left-side "stream" frame. Normally, this frame is responsible only for showing current preview and collected image/movie. I have altered it so that not only it can show images from an alternative source (e.g. a previously-collected image), but also color the displayed image or drawing geometry on top of it. Similarly to existing `ClickService`, all of this functionality is in a separate file, so that there is little modifications to existing structure.

I would really love to merge this code into main, but it has one potential issue: each addition to this `VideoService` adds on some delay between when frame is collected and displayed. This gradually lowers the FPS of the GUI, and as such might be not desirable. As such, I am ready to cut and adapt it so that it is acceptable to anyone. I would love to head some opinions about which features could be potentially valuable to the wider user base.

### The baseline
Current GUI stream frame can only display the preview or the most recent image. On my machine this takes ~6.5 ms, which after adding 50 ms delay results in ~18 fps. Modifying brightness does not seem to affect it, and turning auto-contrast off oddly makes it marginally slower, ~7 ms / 17.5 fps.

### My `VideoService` branch
With all the features implemented but turned off, display time is ~8 ms / 17.5 fps. Modifying brightness (or any other new enhancement) pushes it to ~9 ms / 17 fps. Turning auto-contrast on or off seems to have no effect. My benchmarks are a bit outdated, but in general either applying a custom `matplotlib` gradient to the image or drawing a geometry on top of the image adds ~2 ms on top of that, resulting in ~10 ms / 16.5 fps total.

### The middle way
My modification adds an incredible level of customization to Instamatic. It can help with calibration, show tracked/found crystals, display metadata; it can be even used to display the matplotlib plots that otherwise notoriously break the GUI when run outside the main thread. However, increasing the render time from by a factor of 150% might have more significant effects on lower-end computers. Theoretically GUI runs in a separate thread, but it still takes the same resources as data collection, so excessive modifications to the display might be undesirable. So maybe instead of implementing the whole thing, it would be valuable to include only what is useful - and fast. *And I would love to hear opinions as to what is useful.*
 
[image]

PS: If you find the image **cursed**, I am not sorry 😝.

stefsmeets, 2025-05-08:
Haha, it is extremely cursed. I see how this could be useful though. I have long entertained the idea of something like this, but I had no clear use case for it and my goal for instamatic was to introduce as little lag as possible on the display to not interfere with data collection. The trick is to balance this within the structure of instamatic.

Baharis, 2025-05-08:
I know it is tricky to find the sweet spot, hence this post. At this time I do not need the full system I wrote, I mostly need to be able to provide the display with an alternative image source. Then, using the recently introduced `ClickService`, I can get user input data, provided via clicking on the screen.

@stefsmeets you are saying that your "goal for instamatic was to introduce as little lag as possible on the display to not interfere with _data collection_". Would you then support merging this IF the fps during _data collection_ was still fine? As I mentioned, with complex features turned off, the display is only ~1 ms slower, maybe I can find a way to make this even lower. Then, during data collection (i.e. with drawing off) the fps would be still fine, but some experiments could use this tool to overlay drawings on top of image occasionally, e.g. to plan or mark-up the experiments themselves.

stefsmeets, 2025-05-14:
If this helps you improve your data collection, I'm all for it!

What I meant was that any time spent updating the gui cannot be spend on experiment control. This is because of the way Python does threading. Running *anything* on the gui thread means that events on the data collection thread might delayed. I avoid this by doing the absolute minimum on the gui thread (just update the image and possibly some callback to see if the experiment must be stopped). This is crucial for smooth data collection with predictable timings between frames.