instamatic-dev/instamatic #130: Support numpy >=2

stefsmeets, 2025-05-21:
Numpy is currently fixed to : "numpy >= 1.17.3, <2". This makes instamatic difficult to install with python 3.13, because no binaries are available for numpy<2. For Python 3.13 compatibility, we should update release the upper bound on numpy.

Related commit: https://github.com/instamatic-dev/instamatic/commit/8ab89ec5d7cae06dffcf68048bfa41fae48bb790

Baharis, 2025-05-23:
@stefsmeets Do you have some understanding what is the reason Instamatic is restricted to numpy 2.0 in the first place? I did not have to introduce this restriction in any of my programs, at least not these that I checked.

While we are in the subject, I would also like to float the idea of discontinuing support of Python 3.7 since its official support ended in 23' (and python 3.8 introduced some syntax I like).

stefsmeets, 2025-05-23:
There were a few somewhat non-trivial issues with data type conversions that I was not entirely sure of. So I put an upper limit with the intention of getting back to it when needed.

+1 on dropping support for 3.7 and possibly also 3.8.

Baharis, 2025-05-23:
I don't mind switching to `numpy>2` when I get a chance and keeping an eye for issues. Might take me some time though as my next month is full of travels.

closed 2025-09-19 by Baharis